jascdk/Nilan_Homeassistant #11: MQTT sticky and only update on change

jonaslevring, 2019-08-04:
A suggestion to avoid too frequent and unnecessary updates on MQTT would be to check if a value is changed from last publish and only if so publish the new values.

This concept could be combined with making the messages sticky so the latest values always are available. Even on reconnect to MQTT server.

jascdk, 2019-08-04:
You Think of some kind of retained messages ? Can you make a PR?

jonaslevring, 2019-08-04:
yes.. by sticky i mean retained messages... same concept as suggested works very well in https://github.com/dkjonas/Wavin-AHC-9000-mqtt 
I have looked at a solution, but cannot find the right place to make a decent check, so guess a PR would be pointless.

jascdk, 2019-08-04:
Ahh I see! Well if its purpose is to decrease the stress on the broker it would be fine. I don't have so good coding skills, so I can't make it myself:(

jonaslevring, 2019-08-06:
Would @anderskvist be able to help on this request?

anderskvist, 2019-08-07:
@jonaslevring I might, but I don't really understand why? My broker handles tons of message and this is just a drop in the ocean. Are your broker too busy or running on limited hardware?

jonaslevring, 2019-08-07:
Hardware for broker is limited but not the main concern here.
I'm using this with Homeseer instead of Homeassistant, which qorks fine. But every new message result in new log entry, and creates a new data point entry for visualization.. So eg. a 0/1 bit would create a lot of entries despite it may only change once a year shifting from summer/winter.

I would expect the quality of this project to increase and be even more compatible with other systems that consumes data. If wanted it could be configured whether it should utilize retained messages and only publish on change or always publish.

And by having the suggested change more people could even lower the poll frequency more and get a finer data resolution where it matters without flooding booker or subscribers.

jascdk, 2019-08-07:
Hi @jonaslevring . This project was originally designed for use with OpenHab and with some tweaks on the MQTT side it was going to fit very nicely into Home Assistant. As it is for now, I think the goal here is to strictly use it for the Home Assistant platform. It does not make any issues on this the way it works now. If you want to use it for Home Seer with retained MQTT messages I suggest you fork this repository and make a repo for Home Seer and with the changes in the code there may be. I personally don't have the skills in coding it to fit your needs :/

jonaslevring, 2019-08-07:
Ok.. thank you for your response. Sorry to hear about the decision as i think others could benefit from a great project like this. Especially as it is being maintained.

But anyways I have found a way to solve the issue about retaining messages and only sending updates. It works perfectly without stressing broker or subscriber.

closed 2019-08-07 by jonaslevring

anderskvist, 2019-08-07:
@jonaslevring, isn't it possible to minimize logs in home seer by rotation and compressing them? I personalet prefer the way it is now, because I get a positive confirmation every minute that my system is up and running and I can setup notifications based on that assumption...

jonaslevring, 2019-08-07:
Well.. thats a question of personal preference, consuming systems and partly not filling up queues with irrelevant messages.
With the latest fix for only posting changes i have already received 1100+ messages the last 2 hours. So there are no doubt when the system would't be alive. It could also be supported with an alive-topic for every poll of data.