jascdk/Nilan_Homeassistant #14: Missing landing page to show node status

sveip, 2020-01-16:
What I would like to add, is more info to the root web-page. As of now I cant see any way to get any sensible data from the node if the serial isn't connected (correctly) to the Nilan. It would be great to have a landing page (index.html) that shows serial connectivity status etc.

jascdk, 2020-01-21:
@sveip hi :) have you found out something for this ?

sveip, 2020-01-22:
Not yet, haven't had a chance to look at it yet.

jascdk, 2020-02-04:
> Not yet, haven't had a chance to look at it yet.

Hi @sveip are you on Facebook messenger? :)

sveip, 2020-02-05:
Sorry for the delay, been a busy week. Will try to push a PR tonight.

jascdk, 2020-02-05:
Nice - looking forward to see what you come up with ;)

sveip, 2020-02-05:
Got something that works ok, but not perfect. Seems to be an issue with
closing the client request, will continue tomorrow. :-)

jascdk, 2020-02-06:
Just qurious - is it the landing page your are working on or some kind of wifimanager - integration? :)

sveip, 2020-02-06:
No, what I implementer now is a semi-static page that just shows some
welcome text and the WIFI SSID and MQTT server details. I can push a PR as
is on a branch when I come home, so that you can test it out.