jascdk/Nilan_Homeassistant #16: Link to Danish webshop

JustAnotherUserName87, 2020-02-11:
Hi,

could please you link to the neccessary boards on a Danish webshop? (I am unsure of power between the boards/connecting a 3v board to a 5v board)

Thanks :-)

jascdk, 2020-02-11:
Well - i use this one - it is 3 and 5 volt tolerant so it is no problem hooking it up to a wemos d1 mini or a node mcu. I can’t find it in danish shops :/ https://m.banggood.com/da/3pcs-TTL-to-RS485-Module-Serial-Port-MCU-Automatic-Flow-Control-Module-p-1433019.html?gmcCountry=DK&currency=DKK&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_union&utm_content=xibei&utm_campaign=xibei-ssc-dk-all-j23-da&ad_id=372687366398&gclid=CjwKCAiAvonyBRB7EiwAadauqZwBxFrKgh4GwLMqcU58GAcb33H6ttsoELjDFhl-kVSn9-kIHDEspxoC1psQAvD_BwE

JustAnotherUserName87, 2020-02-13:
Cool! Thank you for you help - is it greatly appreciated :-)

michaelmadsen72, 2021-08-19:
Hi Jacob

Awesome project!

I'm planning to create this project. I have HomeAssistant already set up (RPi) and a Nilan Comfort 300.
I was thinking to buy this one, which is mentioned in the source code: https://www.aliexpress.com/item/32836213346.html?spm=a2g0s.9042311.0.0.27424c4dqnr5i7
As for the Wemos D1 Mini I'm a bit uncertain. I found it here:
https://www.aliexpress.com/item/32831353752.html?spm=a2g0o.productlist.0.0.60c26d82RSEAKJ
but it seems there are 4 different versions to chose from. Do I need any particular one of these?

Nordicfastware, 2021-08-19:
> Hi Jacob
>
> Awesome project!
>
> I'm planning to create this project. I have HomeAssistant already set up (RPi) and a Nilan Comfort 300.
> I was thinking to buy this one, which is mentioned in the source [...]

The TTL to RS485 you linked to is the same I got. It works great.
The Wemods D1, select either the Standard one (D1 Mini) which seems to be V2, or the V3.0. Both would work. I have the V3.0 with my Comfort 300 Top.