jascdk/Nilan_Homeassistant #18: Adding sensor publishing through mqtt in the sketch

Nordicfastware, 2020-02-22:
Hi, 

I have a Nilan 300 Comfort Top and will give your sketch a try as soon as I receive my RS485 to TTL converter part.
However, I would like to also add a CO2 sensor (MH-Z19B) as the Nilan Option for CO2 monitoring is like 3000 kroner or more. So I would like to hook the MH-Z19B on two avoiable GPIOs of the same D1 Mini with your Nilan code on and add whatever is needed in the sketch to handle the new sensor.

Of course there are lots of examples of sketches for just the MH-Z19B on TX/RX or other sensors combined on the D1 mini (like: [https://github.com/minsk-hackerspace/co2Meter](url) but I am not familiar with merging sketches, and also adding items to the MQTT topics to be publish which then would throw out the CO2 readings and temperature from the MH-Z19B.

So i am after a couple of pointers as to how to approach it and perhaps you may have tried something similar and found it unfeasible, in which case I would not waste too much time on that.

If you think this coupling on a single D1 mini is hopeless, then I would have to revert to a separate solution for the sensor, like another D1 Mini, or a simple Sonoff basic with tasmota which is super straight forward. But that seems like a waste of hardware and would require 2 power supplies, cablings etc...

jascdk, 2020-02-22:
Hi @Nordicfastware . Nice that you want to try it out - hope you will enjoy it. Well regarding the MH-Z19 Co2 Sensor I have thought about the same. I have looked at the sensor provided by Nilan, but it is way too expensive! The sensor used by Nilan is called KCD-AN300 made by sensecube (I think):) [https://www.yumpu.com/en/document/view/3339136/kcd-an300-co2-sensor-module](url)

This sensor can be achieved for a lot less money on AliExpress. I have been looking on how it should be attached and it looks like it is feeding the signal through 1 wire to the Nilan PCB like stated and shown here [https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&ved=2ahUKEwjF9KOE2-XnAhWBCOwKHeDaA8wQFjAAegQIAhAB&url=https%3A%2F%2Fwww.nilan.dk%2FAdmin%2FPublic%2FDownload.aspx%3FFile%3DFiles%252FFiler%252FDownload%252FNilanNet%252FWiring%2Bdiagrams%252FDanish%252FCTS602%2BStyring%252FComfort%252FComfort%2B252%2B302%2BTop%252FArkiv%252F7111xx03%2B04%2B06%2B07%2BV1-09%2B180109.pdf&usg=AOvVaw0zUuKCruUTZ82DkTOoyexv](url)

It also looks like it should be connected to the CN11 ports 5 and 6, which is analog ports. That made me wondering, if it would be possible just connecting the MH-Z19 sensor analog out to this port - but Im not sure it would work.  

I have also thought about adding the MH-Z19 to a Wemos running Tasmota for example - this would as you say also be an easy task, but waste of components. If you figure out how to implement this sensors code into this sketch let me know:)

Another add-on I have thought of was implementing a relay into the code, so the user-function (brugervalg in Danish) can be triggered. Anders Kvist almost got it to work, but it is not there quite yet.

Best regards

Jacob

Nordicfastware, 2020-02-23:
Thanks for the feedback. This is all very useful. As I search more details about the analog output from both sensors, I think it won't work, as the MH-Z19B has a 0.4 to 2V proporitional output for the ppm range, while the KCD-AN300 has either 0-5VDC or 0-10VDC or 4-20mA depending on the model of the sensor which goes with the Nilan. So the best is to either purchase the KCD-AN300, along with a 24V power supply converter (rs-15-25, 75kr online), but the issue is that i would not know which voltage output model to order.

Anyways, as discussed the easier is probably to get a tazmotized sonoff basic and hooked up to the MH-Z19, and build some automation in Home assistant instead in case of high CO2.

But first I will try to have the sensor wired to the same wemos as to the RS485 converter.

jascdk, 2020-02-24:
You welcome:) I agree - I don think that this sensor is compatible either :( It would be great, because the right on is damn expensive. I think the best opportunity is to use a tasmotized d1 mini or Sonoff basic as you stated . It would be cool to implement it on the board already used, but how would you think it should be attached? It looks like it uses uart which means RX and TX - they are already used for the Modbus interface:/ I don't know if it is possible to use its analog output .

If you find a solution please make a PR :)

Lutehi-Gregor, 2022-10-21:
Good day
I came across this post via the Google query how to connect Nilan with hassio.
I've been trying to integrate the standard MQTT from GitHub: "Make your Nilan Air ventilating system way more cool" into my hassio installation for a long time.
Unfortunately I keep getting an error message when trying to adjust the config file.
@jascdk, you are the programmer of this integration. Maybe you can give me more information how and where I have to implement these lines of code in hassio.
Many thanks in advance for the support.
With kind regards
Gregory