jascdk/Nilan_Homeassistant #20: Help with implementation

simon-froelich, 2020-03-14:
Hi,

I have tried to make this project work with our Nilan system, but cannot seem to get it working. I am new to a setup like this, so I find troubleshooting a little difficult. Would someone be able to help me?

Here is what I have done: 
Hardware:
- https://www.banggood.com/da/Geekcreit-Doit-NodeMcu-Lua-ESP8266-ESP-12F-WIFI-Development-Board-p-985891.html
-   https://www.banggood.com/da/TTL-to-RS485-Module-Serial-Port-MCU-Automatic-Flow-Control-Module-p-1422016.html

I have connected our Nilan with the TTL-to-RS485 using three of the inner cables from a cat6 ethernet cable. I have used the connections A,B,Shield shown on the last page [here](https://github.com/DanGunvald/NilanModbus/blob/master/MODBUS_CTS-602_2.16-2.19_Installation-and-user-guide.pdf) and connected them to A,B, and what I guess is groud on the converter. The Nilan is also connected to our control panel using some of the same channels.

The TTL-to-RS485 is then connected to the NodeMCU. RXD -> RX, TXD -> TX. 

I have uploaded the code from this repository to the the NodeMCU using Arduino IDE and selecting the board NodeMCU 1.0 (ESP-12E module). My configuration looks like [this](https://pastebin.com/QiaFYN8m).

I can see that the NodeMCU is online on our router and the help page (http://192.168.1.39/help) shows [this](https://pastebin.com/HgWNNBEz).

When I look for MQTT topics in Home Assistant (Developer Tools -> MQTT -> Listen to Topic), where I use Mosquitto, i do not see anything on the topic ventilation/#.

Any idea on that could be wrong? Any help is appreciated.

jascdk, 2020-03-14:
Hi @simon-froelich :) It looks like you have it set up correctly. What I think is wrong is something about the wiring from the modbus pcb to the modbus screw terminals on the Nilan.

You say you use some of the same channels to the modbus pcb as the display (cts602) also uses. 

There are 4 x screw terminals on the Nilan system. These are called A1 B1 and A2 B2. On my system the CTS 602 display is connected to the A1 and B1 . Then there are two terminals left - A2 and B2, which is the ones I used for my modbus pcb. 

Have you connected the CTS 602 and modbus pcb to the same screw terminals? If so - try splitting them up and use both sets of terminals:)

Let me hear from you:)

Best

Jacob

simon-froelich, 2020-03-14:
Thank you for getting back to me! And for sharing your code.

I have attached an image of the Nilan board from before I connected the TTL-to-RS485 (a lot of dust!). The control panel uses, as far as I can see by comparing the wiring and the document, 12V, A2, B2 and ground. I have tried to mark it in the image where I have connected the TTL-to-RS485. 

Does it make sense or should I try and fix an up-to-date photo?

[image]

jascdk, 2020-03-14:
@simon-froelich thanks :D Well yes - then I would say you should use the A and B ports you marked . Then you would use the 2´nd set of modbus ports. Try only using the A and B - you don't need to put on the ground cable from the rs485 module.

jascdk, 2020-03-14:
BTW - what happens if you go to this address in your browser?    

http://192.168.1.39/read/control

Do you get any values from this?

simon-froelich, 2020-03-15:
I have tried to remove the the ground cable. This does not change anything :/

Going to http://192.168.1.39/read/control results in the following, which does not tell me much?

{
  "requestaddress": 1000,
  "requestnum": 8,
  "operation": "read",
  "group": "control"
}

A vacuum would be a great idea and thank you for your help on this

I have attached two pictures of my wiring below. The first one before i removed the ground cable.
[image]
[image]

jascdk, 2020-03-15:
@simon-froelich Ahhhhhh I see what's wrong - you need to feed the TTL board with some juice ie. Power :) Connect it to the 5volt output from the Node MCU;) I recall that the VIN on the nodemcu is 5 v VCC - then use that and a GND pin - the it should work:)

Let me know how it goes;)

simon-froelich, 2020-03-15:
Ahh. I thought that the Modbus signal carried power, but now that you mention it, it makes perfect sense. How did I not think of that..?!

Now I can see the readings from on MQTT 👍 
Next task is to get everything lined up in Home Assistant, but that should be the easy part. I already looked at your configuration and it looks fairly easy. 

Thanks a lot for your help :)

closed 2020-03-15 by simon-froelich

jascdk, 2020-03-15:
No problem - glad you got it to work:)

sadiq81, 2022-11-16:
@simon-froelich or @jascdk it would be really cool if one of you made comprehensive guide with images of alle connections or even a video