jascdk/Nilan_Homeassistant #21: Nilan 300 - CTS 600 support ?

gibman, 2020-03-22:
I have an older nilan 300 EC unit with a CTS 600 board.

Did anyone have any luck connecting an esp8266 to this version of the nilan board ?

Obviously I've detached the wires going to the display and instead rerouted them to the esp modbus device.

Details:
I'm using an esp8266 / wemos which is connected to a modbus to serial unit exactly like this:
https://www.aliexpress.com/item/32828100565.html?spm=a2g0s.9042311.0.0.4d114c4dTKaGuX

the esp is indeed connected to the wifi AP, as I can ping it just fine.
mqtt is also connected, as I can do a test publish call with some static value.

but no modbus values are ever published.

It's pretty hard to debug whats going on as serial debugging is impossible while communicating with the modbus.
I guess I could look into telnet for further debugging.

I could try with another modbus to serial module as well.

med venlig hilsen
gibman :)

jascdk, 2020-03-22:
Hi Gibman. Thanks for you post. Can you please post a picture of your setup and wirings? I think it should work!

MVH

Jacob

gibman, 2020-03-23:
Here are some images of the project.

The wires going out from the NILAN unit is color coded as:

red:12v, not in use for this setup :)
brown: D+/A
yellow: D-/B
blue: GND

this is connected to the socket of the RS485 module, as can be seen on the image.

ESP to RS485 connections are:
ESP gnd => RS485 gnd
ESP 3.3v => RS485 vcc
ESP TxD =>  RS485 RxD
ESP RxD =>  RS485 TxD

the RS485 does have a solid LED light turned on for indicating it has been turned on (VCC)-
the board also has txd and a rxd led.. 
the rxd led does blink here and there. one small quick blink each 2 seconds.. then sometimes longer.
so there is activity going on from the nilan to the module I think.

[image]
[image]
[image]

jascdk, 2020-03-23:
@gibman thanks for the pictures and the description of your connections. Try swapping the rx and tx connections so tx goes to tx and rx to rx. This is how I have mine connected which works. Dont ask me, why they have to be switched around - it is normal that they are crossed:/

gibman, 2020-03-26:
hi there..

tried reversing tx and rx. 
now the rxd led on the rs485 module is totally silent.
So I am convinced that the previous setup is correct in my case atlease.
rx => tx
tx => rx

anyways. I managed to hook up a telnet session in order to further debug it.
it doesnt seem to receive a 0, in the call to ReadModBus
so nothing received :(

'reqtypes rr[] = {reqtemp, reqcontrol, reqoutput, reqspeed, reqalarm, reqinputairtemp, requser, reqdisplay}; // put another register in this line to subscribe
      for (int i = 0; i < (sizeof(rr)/sizeof(rr[0])); i++)
      {
        reqtypes r = rr[i];
        char result = ReadModbus(regaddresses[r], regsizes[r], rsbuffer, regtypes[r] & 1); 
=====>        if (result == 0)
'

gibman, 2020-03-26:
did some more debugging..

it calls into both (L:338 ish):
node.readInputRegisters(addr, sizer);
node.readHoldingRegisters(addr, sizer);

and the result is 226 everytime on each of those 2 calls
so this one equals to false ... result == node.ku8MBSuccess)

here is a list of Modbus Function Codes, Exception Codes
https://4-20ma.io/ModbusMaster/group__constant.html

226 as in my case, equals to 0xE2, which is: ku8MBResponseTimedOut
ModbusMaster response timed out exception.

"The entire response was not received within the timeout period, ModbusMaster::ku8MBResponseTimeout."

jascdk, 2020-03-26:
Hi @gibman . That was strange - I don't have so much knowledge into this debug part sorry. But I came to think of one thing. Have you hooked the RS-485 "on top" of the cts module (display)? As stated in the documents - the cts 600 only have 1 RS 485 port (which is used by the display). My cts 602 however has 2 ports. I use the last port to connect the modbus device to. So I cam to think of if you have doubled the wires on the port of your system? If so - this won't work . Maybe try disconnect the display and see if you get some throughput. All the best / Jacob

jascdk, 2020-03-26:
i also have hooked my Wavin ahc9000 floor heat up to a esp8266 in our house. This unit also uses modbus and have 2 ports, but they are internally connected to each other. This system can be added with a display and I know users have asked - and tried - to hook up a display and the esp on the system, which does not work. I think the same issue can be transferred to your system

jascdk, 2020-03-26:
a second option - have you tried disconnecting the gnd cable to your device? I don't use it on mine.

gibman, 2020-03-26:
hi again..

yes, I knew about the display taking up the only modbus slot that is available.
ofcourse I have retired it :)
so the ESP is the only one wired up here.

Ill try removing the GND cable. This could work.
btw. do you use software or hardware for serial ? in the config.h file ? 

cool regarding wavin. sadly I have the older wavin model prior to ahc9000.
replacing it would cost a ton of money as well as all those wireless thermostats.

Ill report back my findings.

gibman

jascdk, 2020-03-26:
Hi:) Ok - looking forward to hear if you get some results;) For my system I use HARDWARE serial - the SOFTWARE did not work for me. 

Regarding the AHC unit - we did have an old system in our house, which I replaced with the AHC 9000. I was lucky finding a 16 channel - all new model on DBA for about 5 K.

gibman, 2020-03-26:
no succes here either when removing the GND at the nilan end.
tried this in combination with reversing tx and rd again.
didn't help either.

upgrading my nilan board would probably not be worth it (~4000 dkr)

What can be accomplished when using the AHC unit in terms of data and controls ?

Baldorian44, 2020-03-28:
Hello. 
I'm in the same situation as Gibman (or at least I think).  I have a Nilan Comfort 300LR, but with the old (colorless display) which i presume is the CTS 600. But as long as i have the 2 sets of output on the Nilan itself I should be okay right?

I cant get into the circuit board from the Nilan itself without removing the rivets (Nitter in danish), so I would like to avoid that if possible :)

So from the side of my Nilan i have this:
[image]
Which indicates i should have the option to run a 2nd set of wires to control it with.

@jascdk I was thinking of our dear friend https://github.com/dkjonas/Wavin-AHC-9000-mqtt who made the setup for the Wavin system, this setup seems to be more or less identical, except that the protocol here is RS422, however if a chip could be found that would replace it perhaps it could actually work to also power the NodeMCU via the 12V output from the Nilan :)
Did you consider that?

jascdk, 2020-03-28:
Hi @Baldorian44 thanks for your questions :) Ill try to answer ass good I can (im not so much into the technical stuff:/) 

If you have a comfort 300LR as on the picture it looks like the new model and you should have the cts602 panel and also the newer control cts602 control box - later replaced by the touch HMI display. 

As I can see you should be able to adapt this project to your system. 

Is your PCB not accessible from the sidecover where you exchange the filters? It is on mine - right at the left bottom there is a plate which is held with one screw. Under this there is the PCB. I don't have the "breakout" connector like you, so for you it is an easier task by far.

The wavin electrical layout should be able to be adopted to this project also - I have thought of it, but newer realized it :) Dkjonas has some kind of serial swap in his design - where he transmit through other pins. 

Her you use the RX and TX lines on the esp. It is the easiest to do. Get some kind of RS485 board like on the above pictures from @gibman 

You should connect the RS485 to A1 and B1 on your Nilan. The ESP then goes from RX to RX and TX to TX on the RS485 board. Remember to power the RS485 - can be done by using the 3.3 V pin on the ESP (like a Wemos d1 or NodeMCU).

regarding the Half or Full duplex I think this is Half Duplex as it uses 2 wires to transmit and receive through.

@anderskvist has done some experience about the Software Serial, but it never really came to work as expected - it had some issues. 

Let me know how your project is evolving:)

Baldorian44, 2020-03-28:
Hi. Thanks for the reply.
I will have look and see if I can find that screw to take off 😊
Should I be able to see it on the board if it is 600 or 602? Is there a sticker or so?

Yeah as far as I understand after reading a bit more today. Dkjonas uses swap.serial, which just means he uses the other set of UART Tx/Rx pins (on Nodemcu, don't think the Wemos mini has it) . So exactly the same. But when using the MAX3072e you don't have the automatic detecting of transmit/receive as this you use have.
Yeah I did order some of those modules, but that's a 3-6weeks waiting time 😂

Would be nice to make something now as I have parts for the other setup, but will most likely require some rewriting of the code. And yeah time is and issue with kids 😂😂

And yes I think you are right it is half-duplex.

But thanks for sharing this, I'm intreged 😂

sveip, 2020-04-03:
@Baldorian44 I've just started connecting the stuff here as well, and stubled upon this thread that I read some time ago: https://community.openhab.org/t/how-to-connect-nilan-cts602-modbus-interface/71263 
It seems you might be lucky.

For my own case, I opened my Compact S today, and according to this:

[image]
 
I should be ble to connect to the two available ports 2 and 3 marked (RS485 A&B). Is this what you have done @jascdk ? My challenge now is to find a way to rute the cable out of the unit somehow since I cant make any new holes as long as the unit is within warranty at least ;)

[image]

jascdk, 2020-04-04:
@sveip hey Peter :) Yes I use those connectors you mention. Your existing controlbox / display uses the one pair - then just use the remaining pair of ports together with your modbus pcb;) Good luck!

sveip, 2020-04-05:
Thanks @jascdk. The main issue I'm facing is how to get the cable out of the unit. It's quite well sealed, and I guess for a reason. I was hoping to use some unused wires in the existing cables, but it seems they have hidden them well on the outside. The company that built the house have made the inut quite hard to access on the sides and top. How dit you route your cables?

jascdk, 2020-04-05:
@sveip hey :) well my system is only in beta stage so far hehe - so my cable routing is ugly as hell - meaning I use cheap leftover cables running outside underneath the main lid which is to be taken off when changing filters 😃 my goal is to make some kind of routing with cat Ethernet cable out from behind of the unit . There are some unused rubber grommets I can use for this purpose .

Baldorian44, 2020-04-06:
Hey @sveip. Yea its nice to have the connectors on the outside :) but i'm struggling with the RS485 protocol. I have a MAX3072e IC i'm trying to use as my "in-between" device to translate from NodeMcu <--> MAX3072e <--> NilanRS485 port. So far it is not working :)

Good luck with your wiring Sviep.

sveip, 2020-04-09:
@Baldorian44, I would start with the bringup script, and go from there. Any rs485-to-USB dongle should to I recon. You can measure the voltage on the lines to ground to be sure they are in fact connected on the Nilan side.

nagydavid, 2022-07-05:
Hi all, 

Did anyone succeed applying this project without the cts600.

In our [project](https://github.com/laithsaid/nilan_mqtt)  we were able to get all data through a python code, but not the control.

@laithsaid FYI

ronniebach, 2022-10-09:
@gibman  Did you ever manage to get it to work with CTS600 print?

I have similar CTS600 print in my Nilan Comfort 600.
Also I cannot get any modbus communication to work.

So first of all before keep debugging it would be great to know if it is achievable at all

Thanks.

gibman, 2022-10-09:
no sadly I didnt.

but forking out 3000 dkr for a new cts602 would be foolish.

running an esp8266/esphome with temp sensors dht22 located in each chamber.
also a dust particle sensor on the intake.

now I get my stats.´in home assistant.
the esp also has 2 relays.
1; for powering on/off the nilan unit.
2; for starting hi fan speed for eg. humidity high levels.

this is a lot cheaper. although u lose a bit on the flex.

ronniebach, 2022-10-10:
Oh. But did you figure out if the older Cts600 print and the controlling display actually uses Modbus as communication?
Maybe a simpler/reduced set of registeres compared to the cts602 ?

carlhye, 2022-10-27:
Hi Gibman

It's ESPHome and integrated into Home Assistant - no MQTT required.
The module that I have is made by https://github.com/nic6911 and is based on an ESP32. The code have been rewritten to negate the MQTT requirement.

I have both the (old) display and the ESP unit attached - they user different channels A1+B1 and A2+B2.

Actually I'm not 100% sure that I have the CTS600, but since my display states CTS600 and, according to the supplier, that would not work with the CTS602, I assume that I have a CTS600.

I'm going up in the attic to figure it out later today. Will keep you posted.

OB75, 2023-01-15:
Hi Gibman,

I’m interesting in achieving about the same as you I guess, my unit is equipped with CTS600 too, with the limit of only one Modbus slave at a time.

I’m not as far as you, but in the meantime before I start hooking up some HW, I would like to know if you have made any progress?

Are you in possession of the documentation describing a bit about the protocol and connection setup, otherwise I might be able to help with that.

@carlhye  I’m pretty sure your unit is equipped with CTS602, since you have both A1+B1 and A2+B2

pojer, 2023-03-13:
Hello, everybody,
after a long time I tried to integrate my older Nilan VP18K_EK9_M2 ventilation system into Home Assistant. Unfortunately I found out that I have a CTS600 board in it. I tried to read the data using ESP8266+MAX485 with arduino code from https://github.com/jascdk/Nilan_Homeassistant/blob/master/nilan_code/nilan_code.ino according to several procedures, but unsuccessfully.
Has anyone managed to communicate with this board?

Mi2-cz, 2023-09-01:
> Hello, everybody, after a long time I tried to integrate my older Nilan VP18K_EK9_M2 ventilation system into Home Assistant. Unfortunately I found out that I have a CTS600 board in it. I tried to [...]

Hi , i found some artikel on the internet where was wrote that CTS-600 board inside machine as a slave and control panel is master but CTS602 board is master and control panel is slave .that is the reason why 602 works and 600 no.