jascdk/Nilan_Homeassistant #22: USB 485

carlsendk, 2020-04-20:
Hi, 

I had a plan of using USB to rs485 to connect it to rpi with for example home assistant. 

I thought that would be a more simpel setup.

carlsendk, 2020-04-20:
Or this module: https://raspberrypi.dk/en/product/rs485-can-hat-for-raspberry-pi/ 

I know it cost a bid more

jascdk, 2020-04-22:
hi @carlsendk :) Sounds great - it should be possible to use the "native modbus" component in HA to communicate with the system this way. It is more costly as you mention. If you get it to work - please share your experience either here or here https://www.facebook.com/groups/667765647316443/

All the best

/ Jacob

amagr0, 2020-06-05:
Hello guys,

First of all, congrats for this amazing work.

I will acquire a [Nilan 302 Polar ](https://www.nilan.dk/en-gb/frontpage/solutions/domestic-solutions/ventilation/domestic-solutions-ventilation-combi-polar/combi-302-polar) that I'd like to connect to HA.
At this moment my HA system has several temperature/humidity sensors ([BME280](https://www.bosch-sensortec.com/products/environmental-sensors/humidity-sensors-bme280/)) and also some IAQ eCO2 sensors ([iAQ Core C](https://www.sciosense.com/products/environmental-sensors/iaq-core-c/)).
My ideia is to control the Nilan system and the exterior blinds shades based on this sensors to guarantee that the temperature range inside my house is always between 21ºC and 25ºC during all the year.

I'm glad that there is a lot of work done to connect Nilan systems to HA, since I was searching a lot for this. 
I will leave here another hat for RPi that could help you to try to connect directly to RPi.

[How to P2P remote control Modbus-based devices
](https://www.nabto.com/p2p-modbus/?fbclid=IwAR1dzFKV5wXiApgDn3ON-y7OfRSifzu5OS3JtaqFKorwFeGLevIiEwvknZo)

[RS422 / RS485 HAT for Raspberry Pi](https://www.hwhardsoft.de/english/projects/rs485-shield/)

I will give you some news when I start to connect the Nilan to HA. 
(Nilan->RS485->Wemos->MQTT->HA)

Cheers,
André Magro