jascdk/Nilan_Homeassistant #27: Modbus connection failed - WeMos D1 Mini Pro tried both with HW + SW Serial

exetico, 2022-01-09:
Hi,

We can't get a proper connection between RS485 and the WeMos D1 Mini Pro - at least that's what we think.

The RS485 board are connected with RS485 TX to WEMOS TX and RS485 RX to WEMOS RX. We've also tried swapping them. On top of that, we've tried the D1+D2 & D7+D8.

The RS485 board blinks 13 times slowly or so~, and nothing happens for 30 sec.~. Hereafter it starts over again.

But no matter what we do, it's reporting failed connection to modbus. I also expected to see way more traffic on the RX and TX led on the RS485 board...

The RS485 board are powered with 5V (as that's the recommended voltage), but works with the same results on 3,3V, which are also mentioned on the product-listing page.

Could you point us in the right direction? 

<hr>

Here's the current outputs:
```
$curl 192.168.2.117/help
{
  "temp": 0,
  "alarm": 0,
  "time": 0,
  "control": 0,
  "speed": 0,
  "airtemp": 0,
  "airflow": 0,
  "airheat": 0,
  "program": 0,
  "user": 0,
  "user2": 0,
  "info": 0,
  "inputairtemp": 0,
  "app": 0,
  "output": 0,
  "display1": 0,
  "display2": 0,
  "display": 0,
  "operation": "help",
  "group": ""
```

And...

```
$ curl 192.168.2.117/read/control
{
  "status": "Modbus connection failed",
  "requestaddress": 1000,
  "requestnum": 8,
  "operation": "read",
  "group": "control"
```

<hr>

Lastly we've measured the voltage on the PIN's frm the Nilan-system, so there's voltage om them, too. 

The connection between the Nilan-unit and the RS485 are done with GROUND, A1 and B1, connected to A1+, B1- and GROUD. That's matching the manuel, where "ModBus" are mentioned. The display are still connected to A2&B2...

We don't know how to progress from this point. I've checked all issues, but nothing helped us with a final answer. You're more than welcome to mail me, or anything else - Danish is fine... Cause.... Er der ikke kun danskere, i dette repo? 🙈 ...

Nakle, 2022-01-11:
Had the same problem. I was simply missing power to the RS485 TTL-board.

exetico, 2022-01-11:
Hi @Nakle

5V are connected, and it's possible to see a few blinks in RX or TX (depends on how RX and TX are connected, ofc.)

I suggested that GROUND was needed too, but the other part voted what it's not needed for the operation...

Are GROUND needed? If so, I should have hold on to that point 😅..

MURATD0GAN, 2022-01-29:
Hi @exetico,

I have a Comfort CT300 TOP Polar with a similar problem. Did you find a way to make it work?

zinen, 2022-05-29:
This worked for me.  Connected Tx from ESP to Rx on RS485 module and Rx on ESP to Tx on RS485. This is the opposite way then what's in the README.

Then tried `curl 192.168.2.117/read/app` and got a result. 

Then tried some other commands that didn't work. In the end I found that some registers would make modbus return `connection failed` while others succeed.

I went through the whole API to see which registers I could access and modified line 70-133 of [nilan_code.ino](https://github.com/jascdk/Nilan_Homeassistant/blob/master/nilan_code/nilan_code.ino#L70) to fit me needs. My hardware is a Nilan Comfort 300 LR (controller inside ventilation unit is still CTS602) with a touch display (HMI350T).

gjonesy84, 2022-09-22:
Hi Did you ever resolve this issue?

I am also not having much joy connected my node mcu + ttl to rs485 adaptor to my nilan unit.
[image]

gjonesy84, 2022-09-28:
Hi All,

Finally figured out what the issue was. Thought I would share my findings. I have Nilan CT150 connected to a CTS602 controller via modbus 485. I connected to the side plug on the CT150 unit. I used to Modscan to sniff the network and had no data what so ever. This is what my plug looks like.

[image]

And on the back side looks like this!

[image]

I then decided to pull my unit out and check connections and low and behold found that the white and black cables werent even connected.

[image]

I then connected them as below.

[image]

hey presto!

[image]

Sorted!

Hope this helps others!

zinen, 2022-09-28:
Thanks for sharing @gjonesy84. I must admit that I also completely disassembled my unit to find nothing wrong but me trying to connect Tx to Tx and Rx to Rx between Nilan unit and RS485 board which is very wrong ... It should be Tx to Rx and again Rx to Tx.
Glad you got this running.

exetico, 2024-04-28:
We ended up spotting a problem in the wirering to the Nilan, and a few other things. Things finally work at my friends house now. I've also added https://github.com/jascdk/Nilan_Homeassistant/pull/37, with ESP32 support.

closed 2024-04-28 by exetico