jascdk/Nilan_Homeassistant #33: Connecting wires?

michaelmadsen72, 2022-12-28:
Can someone advise if I'm connecting this correctly?
I have a Nilan Comfort 300EC - model 300F with CTS 600 control panel
I didn't connect at the Nilan unit itself, as I figured the same terminals are at the CTS. But it doesn't appear to work, so something is wrong.

[image]
[image]

michaelmadsen72, 2022-12-28:
I think this is the diagram that shows my setup:
https://www.nilan.dk/Files/Files/Dansk/Arkiv/El-diagrammer/CTS600%20Styring_Comfort-300_300-Top_CTS-600_eldiagram.pdf
On page 3, I found this
[image]

But like I said in previous post, it doesn't work, so something is not right.

jascdk, 2023-02-04:
Hi - i dont Think cts 600 is supported :/