jascdk/Nilan_Homeassistant #34: Connection to Modbus ok, but no values in help

Lassenielsen85, 2023-02-09:
Hi Jasc
I've followed your guide and can see that my Wemos d1 mini pro is online when I access the read control page (http://192.168.0.182/read/control). I get the following values:
{
  "status": "Modbus connection OK",
  "Type": 35,
  "RunSet": 1,
  "ModeSet": 1,
  "VentSet": 2,
  "TempSet": 2500,
  "ServiceMode": 0,
  "ServicePct": 5000,
  "Preset": 0,
  "requestaddress": 1000,
  "requestnum": 8,
  "operation": "read",
  "group": "control"
}

When i access the help page (http://192.168.0.182/help) i get no values at all:
{
  "temp": 0,
  "alarm": 0,
  "time": 0,
  "control": 0,
  "speed": 0,
  "airtemp": 0,
  "airflow": 0,
  "airheat": 0,
  "program": 0,
  "user": 0,
  "user2": 0,
  "info": 0,
  "inputairtemp": 0,
  "app": 0,
  "output": 0,
  "display1": 0,
  "display2": 0,
  "display": 0,
  "operation": "help",
  "group": ""
}
However, when i access the read control page (http://192.168.0.182/read/output) i get some values:
{
  "status": "Modbus connection OK",
  "AirFlap": 1,
  "SmokeFlap": 1,
  "BypassOpen": 0,
  "BypassClose": 1,
  "AirCircPump": 0,
  "AirHeatAllow": 0,
  "AirHeat_1": 0,
  "AirHeat_2": 0,
  "AirHeat_3": 0,
  "Compressor": 1,
  "Compressor_2": 0,
  "4WayCool": 0,
  "HotGasHeat": 0,
  "HotGasCool": 0,
  "CondOpen": 1,
  "CondClose": 0,
  "WaterHeat": 0,
  "3WayValve": 0,
  "CenCircPump": 0,
  "CenHeat_1": 0,
  "CenHeat_2": 0,
  "CenHeat_3": 0,
  "CenHeatExt": 1,
  "UserFunc": 0,
  "UserFunc_2": 0,
  "Defrosting": 0,
  "requestaddress": 100,
  "requestnum": 26,
  "operation": "read",
  "group": "output"
}

Do you have a suggestion yo what could be the reason i don't get any temperature readings?

Best regards Lasse

Lassenielsen85, 2023-02-10:
I forgot to mention that I have Nilan 302 polar with CTS602 combined with the HMI350T. From Nilans homepage it says it is the modbus version 22, I guess this could have something to do with me only seing some values.

Lassenielsen85, 2023-02-23:
I found out that it is only the help section that i don't see the values. If I enter any of the read sections (control, output, temp or info) i see all the values listed there.

jascdk, 2023-02-23:
Hey. If I were you - I would strongly recommend using this repo instead https://github.com/Jopand/esphome_components . Much better and stable. Look at the Nilan/Homeassistant community on Facebook for more info on HW etc.

Lassenielsen85, 2023-02-24:
Thanks for the reply! I will definitely look into the esphome_components.